selfcustody/krux-installer #4: KruxInstaller.Setup.0.0.1-alpha-3 Windows Maix Dock M1 installer error

tadeubas, 2023-01-31:
I've tested the installer for Windows `v0.0.1-alpha-3` and got some things to share:

1. It needs to **tell the user to click UNZIP** after downloading and checking signatures from Krux (I've clicked on BACK 🤦‍♂️ and later on I get an error)
2. When executing the program for the second time and selecting the Krux version, I think the **OK button should** be "proceed without download"
3. For **maixpy_dock** the **flash command** NEEDS to use the parameter `-B dan` instead of `-B goE`. Otherwise even with the board connected to the USB, it always gives an error (also the error accumulates on the terminal screen, but for me this is ok)

1)
[image]
[image]

2)
[image]

3)
[image]

qlrd, 2023-01-31:
So there's 3 problems, 2 of them as UX and one is, effectively, a bug, correct?

1. I add this to the branch I'm [working on](https://github.com/qlrd/krux-installer/tree/linux-appimage-wdio);
2. In the above [branch](https://github.com/qlrd/krux-installer/tree/linux-appimage-wdio), I changed the `OK` and `Cancel` buttons to `Download` and `Proceed`. You believe that `OK` (for not download) and `Download` will be better?
3. Bug:
  * About the command change from `-B goE` to `-B dan` I will add this as fast as I can;
  * I will work to remove these accumulation (it's a Vue persistent data question);

tadeubas, 2023-01-31:
Correct!

1. Ok;
2. I would place the `Proceed` first and `Download the file again` later;
3. Yes, it is just that simple parameter change, you can see it on krux code [here](https://github.com/selfcustody/krux/blob/main/krux#L91). IMHO you don't need to remove these accumulation, but do as you wish 😄

qlrd, 2023-01-31:
2. " place the Proceed first and Download the file again later " fixed on branch `linux-appimage-wdio`.

Check the commit [8acade0](https://github.com/qlrd/krux-installer/commit/8acade0d1162b9ef6aa805163a370b82aa503b94) and its [e2e test](https://github.com/qlrd/krux-installer/actions/runs/4059221895/jobs/6987036330) to verify.

Once i finish all tests on linux, i will test them on windows (an maybe a virtualized mac on github).

qlrd, 2023-02-03:
1. Created a warning message from your request;

I don't know if this positioning of the message on the left side and buttons on the right side is adequate. What do you think?
[image]

tadeubas, 2023-02-04:
I think it is awesome, the text on left aligned with the buttons on the right makes the user read and understand what to do next... I would only change "You will need to unzip this release before flash" to "You need to UNZIP this release before flash" just to make sure the user follow the correct path.

One question I have is the case of the signature verification fail... maybe hide the Unzip button? Or place another text in saying something in the WARN:?

qlrd, 2023-02-19:
Sorry for late response;

> One question I have is the case of the signature verification fail... 

If signature verification response fail, the user will receive an "alert" message and the page goes to main page; then you can download again the release.

qlrd, 2023-04-26:
Fixed on  [4996231](https://github.com/selfcustody/krux-installer/commit/49962314074aecaaaa951c40d73f3d8491381c7a) and [1a7c6ba](https://github.com/selfcustody/krux-installer/commit/1a7c6bac7379a706f816738e03a1e1573f3eabf0)

closed 2023-04-26 by qlrd