selfcustody/krux-installer #44: Add Krux QR code encrypt / decrypt to the GUI

tadeubas, 2024-02-21:
The full script is that needs to interact with the GUI: https://gist.github.com/odudex/e86fb4d97cc2c161803ad01f7b43e237

Doc about Krux Encrypted Mnemonics: https://selfcustody.github.io/krux/getting-started/features/encrypted-mnemonics/

qlrd, 2024-08-21:
I'm reviewing all the old stuff that I haven't resolved... 

I don't remember the details and I'll try to infer what you meant.

Is your proposal to capture an encrypted qrcode on the device, decode it and show the contents in the installer?

tadeubas, 2024-10-03:
Sorry for the late reply. The idea is to be able to decrypt a Krux QRCode or an encrypted mnemonic stored on the SD card. To be able to do what a Krux device is capable of doing, but without the Krux device!

This is important because the user may have an encrypted backup (QR or saved on the SD), but no longer have the device to decrypt it (this will also address a workaround for a rare bug in `v23.09.0` that may require brute-forcing to recover the mnemonic).

qlrd, 2026-08-15:
I was really thinking deeply on this matter during those month. "That" triggered more event thoughts.

TBH,to have a `decrypt-kef` on installer go beyonnds the facilitation to recover secrets in a calamity situation: i think it could be not proper to use installer as a `suite-tool` and this could be done properly in a dedicate CLI to avoid any theoritical harm -- IMO this isn't a too "plumb work" like installer was).

Since this i will close this issue not because i do not agree, but IMO is better for comunitty to have a dedicated too for `decrypter-kef` in a way that should be secure to who uses it. Since them, i will work with some sketches, but i invite you guys to help assemble the party.

closed 2026-08-15 by qlrd