selfcustody/krux-installer #115: [Bug] krux-installer not starting on Fedora 41 with Gnome (Wayland)

ciori, 2024-12-03:
I installed krux-installer (version v0.0.20-beta) using the rpm package on a Fedora 41 system (with Gnome and Wayland), but when I try to open the app it quits almost instantly and it gives this error when I launch it from a terminal:
```
...
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [KruxInstallerApp          ] Adding resources from /tmp/_MEIYUGphJ
[INFO   ] [KruxInstallerApp          ] Registering assets path=/tmp/_MEIYUGphJ/assets
[INFO   ] [KruxInstallerApp          ] <kivy.config.ConfigParser object at 0x7fafea58aab0>.lang=en_US.UTF-8
[INFO   ] [GL          ] NPOT texture support is available
 Traceback (most recent call last):
   File "krux-installer.py", line 29, in <module>
   File "kivy/app.py", line 955, in run
   File "kivy/app.py", line 925, in _run_prepare
   File "src/app/__init__.py", line 77, in build
     ----------------------
                            
   File "src/app/screens/ask_permission_dialout_screen.py", line 52, in __init__
   File "src/app/screens/ask_permission_dialout_screen.py", line 153, in detect_usermod_bin
   File "src/app/screens/base_screen.py", line 394, in redirect_exception
 AttributeError: 'NoneType' object has no attribute 'get_screen'
[PYI-35402:ERROR] Failed to execute script 'krux-installer' due to unhandled exception!
```
(I already restarted the system after installing the package)

While version v0.0.20-alpha-3 is working.

qlrd, 2024-12-04:
Hi @ciori, thanks for the report.

I will try to investigate more and give you some answer.

ciori, 2024-12-04:
Thank you, this is the output: `/usr/sbin/usermod`

qlrd, 2024-12-04:
> Thank you, this is the output: `/usr/sbin/usermod`

Strange. It should got the operational system.

What is the output of `cat /etc/os-release`? (We need only the `ID_LIKE`, or `ID` part). 

You can do something like:

- `cat /etc/os-release | grep ID_LIKE`
- `cat /etc/os-release | grep ID`

ciori, 2024-12-05:
`ID_LIKE` is not present, while `ID` is simply `fedora`

qlrd, 2024-12-05:
Gotcha, so the problem is [this one](https://github.com/selfcustody/krux-installer/blob/13570ecc5360ea8f9d91df23b24abd958baf6e6a/src/app/screens/ask_permission_dialout_screen.py#L133) . 

I suposed that "fedora" would be a `ID_LIKE=rhel`, since it is derived from RedHat.

But what is really strange to me is that was suposed this never meant to be called. I mean, a `post-install` section on `install` [script](https://github.com/selfcustody/krux-installer/blob/13570ecc5360ea8f9d91df23b24abd958baf6e6a/.ci/create-rpm.sh#L191C2-L191C6) should have called the `usermod -a -G dialout $SUDO_USER`.

Could you give another hint about the output of `groups` command and check if `dialout` is present?

Just to

ciori, 2024-12-05:
Yep, dialout is present on my user's groups: `ciori wheel dialout`.

I tried adding `ID_LIKE=rhel` inside `/etc/os-release` and the krux-installer now starts without errors, but I don't know whether that `rhel` value is correct/should be there for a fedora system (I don't think it should, but I am not 100% sure).

qlrd, 2024-12-06:
Gladly to know that a hard-code fix it, but... like [this thread on reddit](https://www.reddit.com/r/linuxquestions/comments/ouinvo/changing_values_in_etcosrelease/)

> /etc/os-release will be a symlink to /usr/lib/os-release at the moment, but it is valid to replace that symlink with a regular file. But... I wouldn't do that. The whole point of the file is that programs can use it to know what operating system you're running, and adapt their operations accordingly. If you put incorrect values in it you're only going to break those programs.

As workaround to know if it fix the installer bug, i think it's ok. But i believe that, in long term, isnt the best thing to do.

I can try a fix with a little patch:

```python
            # For Fedora, to fix issue #115
            # see https://github.com/selfcustody/krux-installer/issues/115
            elif os_data.get("ID") in ("fedora"):
                bin_path = "/usr/sbin/usermod"
```

I think this will be enough.

Then it will be needed to run from source on Fedora with `poetry run poe dev`

qlrd, 2024-12-06:
@ciori, can you test with built artifact `rpm` [here](https://github.com/qlrd/krux-installer/actions/runs/12203636016)?

ciori, 2024-12-06:
Yes, here is the result:

By running `krux-installer` from the terminal, the app starts and displays this message:
[image]
Then if I press `Allow`, an error is displayed:
[image]

This is the corresponding log from the terminal:
```
...
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [KruxInstallerApp          ] Adding resources from /tmp/_MEIWmaulx
[INFO   ] [KruxInstallerApp          ] Registering assets path=/tmp/_MEIWmaulx/assets
[INFO   ] [KruxInstallerApp          ] <kivy.config.ConfigParser object at 0x7f0f7742a090>.lang=en_US.UTF-8
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [Base        ] Start application main loop
Fedora Linux 41 (Workstation Edition) not supported
[ERROR  ] [ErrorScreen          ] Fedora Linux 41 (Workstation Edition) not supported
_path_normpath: path should be string, bytes or os.PathLike, not NoneType
[ERROR  ] [ErrorScreen          ] _path_normpath: path should be string, bytes or os.PathLike, not NoneType
[INFO   ] [Base        ] Leaving application in progress...
```

I tried by already having `dialout` on my user's groups and without the `ID_LIKE=rhel` parameter on `/etc/os-release`. I also tried after removing the rpm and rebooting the PC multiple times just to make sure, but the app still shows that error.

I also tried by setting `ID_LIKE=rhel` again and it starts without errors and without the intial windows from the screenshots above.

Just my two cents:
I see the line `elif os_data.get("ID") in ("fedora",):` was added, but shouldn't it be `elif "ID" in os_data and "fedora" in os_data["ID"]:` instead, similar to the rhel one? Also, I am not sure about this either, but could it be that this happens because the string `ID` is contained on multiple lines inside `/etc/os-release` making the check not quite working and giving `None` as result, while `ID_LIKE` appears only once and therefore is working?

Hope this can help.

qlrd, 2024-12-07:
Thanks @ciori, will try that with a fedora VM and hope your cents work :)

qlrd, 2024-12-07:
@ciori, i think it worked (at least on VM):

[image]

Can you try with this [build](https://github.com/qlrd/krux-installer/actions/runs/12216277841)?

What was fixed: both solutions (mine and yours work). But the mistake was that i placed, in the previous commit, in the wrong file.

ciori, 2024-12-08:
Yes, can confirm it is now working as expected, thank you.

qlrd, 2024-12-08:
Great!

Thanks for help @ciori 

I wont close this issue now since it will be closed with the #109

ciori, 2024-12-08:
I only tested whether the app was starting, I will be able to test the flash procedure only in the following days.

I'll let you know.

ciori, 2024-12-15:
Hi, sorry for the waiting.

I have tested the flashing procedure and I encountered an issue: basically the application gives an error while trying to connect to the device I am using (Maix Amigo), these are the logs from terminal:
```
[INFO   ] [FlashScreen          ] [INFO] COM Port Selected Manually:  /dev/ttyUSB0 
[INFO   ] [FlashScreen          ] [INFO] Default baudrate is 115200 , later it may be changed to the value you set. 
[INFO   ] [FlashScreen          ] [INFO] Trying to Enter K210 ROM ISP Mode... 
[INFO   ] [FlashScreen          ] *
[INFO   ] [FlashScreen          ] 
[INFO   ] [FlashScreen          ] Greeting fail, check serial port (SLIP receive timeout (wait frame start)) for /dev/ttyUSB0
Flash failed:
Greeting fail, check serial port (SLIP receive timeout (wait frame start)) for /dev/ttyUSB0

Ensure that you have selected the correct device and that your computer has successfully detected it.

 Exception in threading.excepthook:
 Traceback (most recent call last):
   File "threading.py", line 1381, in invoke_excepthook
   File "src/app/screens/flash_screen.py", line 210, in hook
   File "src/app/screens/base_screen.py", line 414, in redirect_exception
   File "src/app/screens/base_screen.py", line 132, in set_screen
   File "kivy/properties.pyx", line 520, in kivy.properties.Property.__set__
   File "kivy/properties.pyx", line 567, in kivy.properties.Property.set
   File "kivy/properties.pyx", line 606, in kivy.properties.Property._dispatch
   File "kivy/_event.pyx", line 1307, in kivy._event.EventObservers.dispatch
   File "kivy/_event.pyx", line 1213, in kivy._event.EventObservers._dispatch
   File "kivy/uix/screenmanager.py", line 1063, in on_current
   File "kivy/uix/screenmanager.py", line 372, in start
   File "kivy/uix/screenmanager.py", line 397, in add_screen
   File "kivy/uix/screenmanager.py", line 1041, in real_add_widget
   File "kivy/uix/floatlayout.py", line 140, in add_widget
   File "kivy/uix/layout.py", line 97, in add_widget
   File "kivy/uix/widget.py", line 673, in add_widget
   File "kivy/graphics/instructions.pyx", line 659, in kivy.graphics.instructions.Canvas.add
   File "kivy/graphics/instructions.pyx", line 670, in kivy.graphics.instructions.Canvas.add
   File "kivy/graphics/instructions.pyx", line 88, in kivy.graphics.instructions.Instruction.flag_data_update
 TypeError: Cannot change graphics instruction outside the main Kivy thread
[ERROR  ] [ErrorScreen          ] Flash failed:
Greeting fail, check serial port (SLIP receive timeout (wait frame start)) for /dev/ttyUSB0

Ensure that you have selected the correct device and that your computer has successfully detected it.
```

I tried different usb cables and even computers (only Fedora though), but it was still giving the same error, so then I tried to better understand how this serial port connection should work.

When I connect the device to the computer I can see it binds (I am not sure this is the correct term) both `/dev/ttyUSB0` and `/dev/ttyUSB1`, but if I try to access those serial ports with programs like `screen` or `minicom` I can see some strings referencing the Amigo device only while using the USB1 port.

After some tests I ended up with this workaround that made the flashing procedure work successfully:

1. Start with the device powered off
2. Insert the device in the computer using the usb cable (simple USB-C to USB-A cable) and wait for it to turn on
3. Delete `/dev/ttyUSB0` and create a new symbolic link to the working `/dev/ttyUSB1`
  ```
  sudo rm -rf /dev/ttyUSB0
  sudo ln -s /dev/ttyUSB1 /dev/ttyUSB0
  ```
4. Open `krux-installer` and flash/wipe the device
5. The flashing procedure is now using the USB1 serial port and it completes successfully

So the idea was to force the app to use the other serial port, which apparently is the only one working.

I don't know if this depends from the linux system, the cables I used, the specific amigo device or what, but in the end the flashing procedure only works while using `/dev/ttyUSB1`.

This error might not even be related to this build or even the fact that I am using Fedora (as was the initial problem of this issue).

PS: no problems with the airgapped flashing option using an SD card.

Hope this can help.

qlrd, 2024-12-15:
> I have tested the flashing procedure and I encountered an issue: basically the application gives an error while trying to connect to the device I am using (Maix Amigo)

Reproduced here.

> I tried different usb cables and even computers (only Fedora though), but it was still giving the same error, so then I tried to better understand how this serial port connection should work.

I investigated what was and found that wasnt a serial port connection issue (I accidentally knocked my Amigo to the ground and thought the problem was the device until you reproduced it - fortunately the problem was in the software). 

> So the idea was to force the app to use the other serial port, which apparently is the only one working.

This procedure is already implemented. 

> This error might not even be related to this build or even the fact that I am using Fedora (as was the initial problem of this issue).

In summary, the issue was in the middle of the procedure, when i fixed another bug (see #117): before the 2nd flash/wipe try occurs, some KTool's logs are propagated (that it is used to track, in realtime, what it happens), including a one `"Cancel: Greeting fail"`.

It happened that its propagated the 1st's port `"Cancel: Greeting fail" before the 2nd's port try occurs, stopping the entire procedure. Placing the propagation after 2nd try fixed it ( 14ecf405), at least, on my machine.

Please test this [new build](https://github.com/qlrd/krux-installer/actions/runs/12342485711)

ciori, 2024-12-16:
> Please test this [new build](https://github.com/qlrd/krux-installer/actions/runs/12342485711)

Yes it works, thank you.

ciori, 2024-12-16:
Glad to be of help

closed 2024-12-20 by qlrd