selfcustody/krux-installer #127: [Bug] Flashing on macOS Blank screen or error

joksa, 2025-01-23:
#### Describe the bug

All went fine until the last step.

Flash With .... shows an error in the 6th image
AirGapped ... shows a blank screen, the last image

NOTE: I made an update using the manual method, copied the firmware to the SD Card, and plugged it into Yahboom

Prepared USB Flash Drive
[image]

[image]
[image]
[image]
[image]
[image]
[image]

qlrd, 2025-01-23:
Thanks for reporting @joksa, can you provide more information?

* Which OS are you running, MacOS Intel or MacOS M1/M2/M3?

* Which version of Installer you're running?

> Flash With .... shows an error in the 6th image

About this error, it indicate that the device isn't recognized by system. So i would like to ask if you plugged the device directly to the port or used a hub (on MacOS M1/M2/M3 is necessary to use a USB docker).

> AirGapped ... shows a blank screen, the last image

When i developed, i caught sometimes that error (but it was supposed to raise an exception on screen, so i will ask again, which version of installer you're using?)

About the error itself, it happened to me when the SDCard isn't recognized by system. 

I needed then, after plug SDCard, to wait a little for system detect it. Generally a window popup appear within the SDCard root directory after recognized and an Icon appears at the side of Finder window with a "eject" image.

So i will ask: the SDCard it was really detected? If no, it's a SDCard issue. If yes i will ask for run installer from terminal, re-do the steps and catch some logs:

```bash
$> /Applications/krux-installer.app/Contents/MacOS/krux-installer
```

Or:

```bash
LOGLEVEL=debug /Applications/krux-installer.app/Contents/MacOS/krux-installer
```

CAUTION: this last one will show a lot of messages and some verification procedures will take longer.

Sorry for a verbose message, but i need them.

joksa, 2025-01-23:
OS: macOS 15.1.1 on M1 Pro
KruxInstaller: krux-installer_0.0.20-beta_arm64
- Device Directly plugged in via USB cable
- SDCard is formatted and recognized (the first image) I used it later to manually copy the firmware by hand

LOGS (1st call):

[INFO   ] [Logger      ] Record log in /Users/joksa/.kivy/logs/kivy_25-01-23_3.txt
[INFO   ] [Kivy        ] v2.3.0
[INFO   ] [Kivy        ] Installed at "/var/folders/sz/pttxy1nd2s7dgfww1kxpfth80000gn/T/_MEI3Z7spA/kivy/__init__.pyc"
[INFO   ] [Python      ] v3.12.7 (v3.12.7:0b05ead877f, Sep 30 2024, 23:18:00) [Clang 13.0.0 (clang-1300.0.29.30)]
[INFO   ] [Python      ] Interpreter at "/Applications/krux-installer.app/Contents/MacOS/krux-installer"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[INFO   ] [Factory     ] 195 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_imageio, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <sdl2>
[INFO   ] [GL          ] OpenGL version <b'2.1 Metal - 89.3'>
[INFO   ] [GL          ] OpenGL vendor <b'Apple'>
[INFO   ] [GL          ] OpenGL renderer <b'Apple M1 Pro'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 1
[INFO   ] [GL          ] Shading version <b'1.20'>
[INFO   ] [GL          ] Texture max size <16384>
[INFO   ] [GL          ] Texture max units <16>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [KruxInstallerApp          ] Adding resources from /var/folders/sz/pttxy1nd2s7dgfww1kxpfth80000gn/T/_MEI3Z7spA
[INFO   ] [KruxInstallerApp          ] Registering assets path=/var/folders/sz/pttxy1nd2s7dgfww1kxpfth80000gn/T/_MEI3Z7spA/assets
[WARNING] [KruxInstallerApp          ] None not supported. Default <kivy.config.ConfigParser object at 0x106398c50>.lang=en_US.UTF-8
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [GL          ] BGRA texture support is available
[INFO   ] [Base        ] Start application main loop
[DEBUG  ] [Starting new HTTPS connection (1)] api.github.com:443
[DEBUG  ] [https          ]//api.github.com:443 "GET /repos/selfcustody/krux/releases HTTP/1.1" 200 None
2025-01-23 11:34:52.691 krux-installer[7976:193282] +[IMKClient subclass]: chose IMKClient_Modern
2025-01-23 11:34:52.691 krux-installer[7976:193282] +[IMKInputSession subclass]: chose IMKInputSession_Modern
[WARNING] [Label       ] pop style stack without push
[ERROR  ] [FlashScreen          ]   File "src/utils/flasher/base_flasher.py", line 93, in port
StopIteration

 Exception in threading.excepthook:
 Traceback (most recent call last):
   File "threading.py", line 1381, in invoke_excepthook
   File "src/app/screens/flash_screen.py", line 190, in hook
   File "src/app/screens/base_screen.py", line 403, in redirect_exception
   File "src/app/screens/base_screen.py", line 132, in set_screen
   File "kivy/properties.pyx", line 520, in kivy.properties.Property.__set__
   File "kivy/properties.pyx", line 567, in kivy.properties.Property.set
   File "kivy/properties.pyx", line 606, in kivy.properties.Property._dispatch
   File "kivy/_event.pyx", line 1307, in kivy._event.EventObservers.dispatch
   File "kivy/_event.pyx", line 1213, in kivy._event.EventObservers._dispatch
   File "kivy/uix/screenmanager.py", line 1063, in on_current
   File "kivy/uix/screenmanager.py", line 372, in start
   File "kivy/uix/screenmanager.py", line 397, in add_screen
   File "kivy/uix/screenmanager.py", line 1041, in real_add_widget
   File "kivy/uix/floatlayout.py", line 140, in add_widget
   File "kivy/uix/layout.py", line 97, in add_widget
   File "kivy/uix/widget.py", line 673, in add_widget
   File "kivy/graphics/instructions.pyx", line 659, in kivy.graphics.instructions.Canvas.add
   File "kivy/graphics/instructions.pyx", line 670, in kivy.graphics.instructions.Canvas.add
   File "kivy/graphics/instructions.pyx", line 88, in kivy.graphics.instructions.Instruction.flag_data_update
 TypeError: Cannot change graphics instruction outside the main Kivy thread
[ERROR  ] [ErrorScreen          ] Flash failed:   File "src/utils/flasher/base_flasher.py", line 93, in port
StopIteration

qlrd, 2025-01-23:
> KruxInstaller: krux-installer_0.0.20-beta_arm64

This beta version really had a bug. It was suposed to be fixed on [0.0.20 release](https://github.com/selfcustody/krux-installer/releases/tag/v0.0.20). Please update it and test again.

> Device Directly plugged in via USB cable

We recomend, in mac, use a docker hub like [this](https://www.amazon.com/USB-Hub-Multiport-Adapter-Compatible/dp/B08397MVWD/). I don't know why, direct connection sometimes work and sometimes do not work 🤕 .

Like we talked about in a DM, i will document here to double-check for permissions on removable drives.

joksa, 2025-01-23:
The issue has been resolved.
The firmware was updated with a USB cable using the same configuration.
SDCard update is still an issue but not a big deal ATM.

closed 2025-01-23 by joksa